JohnDoneth/hd44780-driver #4: Non Blocking API

jonwingfield, 2018-10-24:
Can you share your thoughts around the non-blocking API? I'd be willing to take a stab at it, but I'm not sure what you had in mind.

JohnDoneth, 2018-10-25:
Well the API I had in mind would be to spawn a future when issuing a command as not to block the main thread. Then wait for the busy signal from the LCD to end and use that as the indicator that the future should be ended when polling. 

I know the `nb` crate has some useful constructs already that could be used in the process of developing the API surface, but I have not looked into it recently.

xoviat, 2021-01-22:
I'm willing to implement this. It's a simple matter of accepting an async delay, and then modifying the api to "async",.

zacck, 2021-03-01:
Hey @xoviat  would it be possible to know if any progress was made on this, considering using this library with rtic.rs

xoviat, 2021-03-01:
The problem is that there are no async traits. rust-embedded/embedded-hal#206 was submitted, but that went nowhere. I had considered just adding a dependency on embassy (the async library that I'm using), but I haven't done that yet. The problem with RTIC IMO is there's no encapsulation, which is why I like embassy better.

zacck, 2021-03-01:
@xoviat  and if you use embassy, how would you implement the solution? I'm a tad new to rust trying to figure out how to get past this more than keen to try do the work

xoviat, 2021-03-01:
@zacck can you join https://app.element.io/#/room/#rust-embedded:matrix.org so that we can discuss this?

closed 2021-03-03 by JohnDoneth

zacck, 2021-03-11:
Hey @JohnDoneth  would you please make a new release for this, for some reason I cant reach the masync module at this moment using this as github dependency.

JohnDoneth, 2021-03-11:
@zacck Sure! I will prioritize getting a release out. In the meantime did you add `features = ["async"]` to your git dependency? The async feature is not enabled by default.

zacck, 2021-03-11:
@JohnDoneth  I did not do that, its likely what I need! Still learning

PR #27 (2021-03-03): add async functionality
This PR adds async functionality. However, it removes the blocking functionality. This is really more of a conversation starter on how both blocking and async should be implemented. Should I copy the async into a new folder, and have them side-by-side. Should there be a configurable feature?

Closes #4.