JohnDoneth/hd44780-driver #12: Add I2C support

nils-van-zuijlen, 2019-11-13:
I2C backpacks are available for HD44780 LCDs. They are mostly found in projects where the micro-controller does not have many pins or where an I2C bus is already in use.

A schematic for these backpacks can be found at https://www.electroschematics.com/arduino-i2c-lcd-backpack-introductory-tutorial/

It could be implemented by using a FourBitBus over I2C and by adding other controls for the RS, RW, E and BT (backlight) bits.

An Arduino library for driving LCD modules using these backpacks is available at https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library for reference

JohnDoneth, 2019-11-17:
Closed by #13

closed 2019-11-17 by JohnDoneth

mpbraendli, 2020-05-18:
Very happy to have discovered this! The fact that the i2c backpacks are supported is however not mentioned in the README, and it took me some searching to find out. Looking forward to trying out your library, many thanks!

JohnDoneth, 2020-05-18:
@mpbraendli Thanks for the feedback! I'll put that in the README for future users.

mpbraendli, 2020-06-06:
I found out that I had a different i2c backpack, with an MCP23008 expander (adafruit sells those).
I've added support [here](https://github.com/mpbraendli/hd44780-driver/commit/52f4bcdef41d4dedef1a7502231ac961f3ee1dd1) and I feel I'm close but not quite there yet:

[image]

This is supposed to show the customary "hello world"...

Would you perchance have an idea that could help me further? Did you ever see similar behaviour with your screens?

One thing I noticed:
```python
In [1]: bin(ord('d'))                                                                                                                                                                                                                                                   
Out[1]: '0b1100100'

In [2]: bin(ord('l'))                                                                                                                                                                                                                                                   
Out[2]: '0b1101100'
```

JohnDoneth, 2020-06-06:
@mpbraendli Interesting! I would try playing with the timing and seeing if that fixes it.

mpbraendli, 2020-06-13:
There was an issue with my wiring that led to this half-working/half-broken state. It works fine now.

I've also improved documentation a bit, if you wish to take these modifications you can pull them from https://github.com/mpbraendli/hd44780-driver

Thanks again!

piegamesde, 2020-08-27:
Please make a new release so that the i2c support actually shows up in the docs on crates.io

JohnDoneth, 2020-08-27:
@piegamesde Thanks for the reminder, you can watch this issue #25 for the release.