JohnDoneth/hd44780-driver #19: Arduino UNO Example

JohnDoneth, 2020-07-25:
Now that Rust has nightly support for AVR it would be neat to have an example for this crate using [avr-hal](https://github.com/Rahix/avr-hal) for the ubiquitous Arduino UNO.

ColinTimBarndt, 2024-09-26:
In #54, I added an example for the Arduino Nano, but it's general enough to work on any Arduino as it's using [arduino-hal], which is part of the same project and depends on avr-hal.

[image]

[arduino-hal]: https://github.com/Rahix/avr-hal/tree/main/arduino-hal

mrbell321, 2025-02-28:
@ColinTimBarndt it looks like the nano example in the repo is just a Blinky program and does not use the hd44780 driver?