JohnDoneth/hd44780-driver #25: Release 0.4.0 on crates.io

JohnDoneth, 2020-08-27:


OliverEvans96, 2020-09-15:
Hi there,

Thanks for this great package.

It looks like there hasn't been a release in over 1.5 years, it would be great to have a new version pushed to crates.io. If 0.4.0 still requires substantial work, I wonder if you would consider publishing 0.3.1 as-is?

I'm using avr-hal (arduino), which uses the digital::v2 traits from embedded-hal. Your master branch has been updated, but 0.3.0 still uses the old v1 traits, which makes it incompatible with the latest avr-hal.

Thus, to fix #19 would require one of the following:
1. Release a new version to crates.io with this update
2. Use `hd44780-driver/master` in the arduino uno example
3. Use an outdated version of `avr-hal` in the arduino uno example

It seems to me that 1. is the most beginner-friendly solution. However, I have managed to use approach 2. to get a minimal example I can share.

All the best,
Oliver

OliverEvans96, 2020-09-15:
Actually, I realize now that `avr-hal` is entirely unpublished on crates.io, so I guess one more git dependency isn't the end of the world :rofl:

JohnDoneth, 2020-09-15:
I've been meaning to get around to this. I just released 0.4.0, hopefully that relieves at least one of your issues :smile:.

closed 2020-09-15 by JohnDoneth