JohnDoneth/hd44780-driver #36: Suggestions on how to handle boards with slow RTCs?

anahata0108, 2021-05-19:
Hello!

I'm currently playing with driving an HD44780 with a SiFive HiFive1 RevB. Its RTC is only 32_786 Hz, and so is too slow for a `DelayUs` impl. I have a fork where I've changed all the `DelayUs<u16>` trait bounds to `DelayMs<u16>` and changed all the `delay.delay_us(100)` calls to `delay.delay_ms(1_u8)` calls, which seems to work. This is, of course, a terrible hack, so I'm curious what your thoughts are for supporting boards with slow RTCs. 

In terms of prior art, there's [another fork](https://github.com/mpbraendli/hd44780-driver/commit/67c318c2e3697dfbf17fc803ea66e40d1caf5743) which uses a separate bus module for the board they're working on, relying on I2C.

I'd love to hear your thoughts on this.

anahata0108, 2021-05-20:
I experimented some more last night after filing this issue, which revealed some oddities. First, occasionally when writing data to the LCD, a character will not render properly. One of the specific ways this manifests is that sometimes a 'z' will render as an 'r'. Looking at the [datasheet provided by Adafruit](https://cdn-shop.adafruit.com/datasheets/HD44780.pdf) (pp 17-18), these are represented by `0b0111_0010` and `0b0111_1010` respectively on the HD44780. The code I'm using is [anahata0108/lcd_fiddle](https://github.com/anahata0108/lcd_fiddle) if you want to take a look at what I'm doing. 

I have some other boards to play with (like an Uno) and I'll see if I can reproduce the same bug with them. The HiFive1 RevB is known for having a slow RTC, and some other oddities, so I'll do some more digging.

ColinTimBarndt, 2024-09-30:
Delay implements DelayUs now in version 0.11.0: https://docs.rs/e310x-hal/0.11.0/e310x_hal/delay/struct.Delay.html#impl-DelayUs%3Cu8%3E-for-Delay