JohnDoneth/hd44780-driver #38: Implement two-dimensional cursor addresses.

rene0, 2022-05-08:
Currently the `set_cursor_pos()` method takes a one-dimensional address to move the cursor. The article at [1] shows how to calculate this address from an (x,y) coordinate pair for various display sizes. Note that the example
```
set_cursor_pos(40, &mut delay); // move to line 2
```
only works by accident, because "The LCD controller automatically accounts for the gap in addresses and after storing an ASCII code in address 27h it puts the next code in address 40h.   Similarly it increments from address 67h back to 00h." ([1], just before "40x2 LCD")

An implementation of this calculation is done in [2] (copied from [3]), which I tested on a 20x4 display. Perhaps the `panic()` does not make sense for the library, and perhaps the constructors could take the number of columns and rows into account?

[1] https://web.alfredstate.edu/faculty/weimandn/lcd/lcd_addressing/lcd_addressing_index.html
[2] https://github.com/rene0/rp-rs-radio-clock/blob/hardware-tests/test-display/src/main.rs#L29
[3] https://github.com/dhylands/python_lcd/blob/master/lcd/lcd_api.py#L122

rene0, 2022-05-23:
[2] has been replaced by https://github.com/rene0/rp-rs-radio-clock/blob/1d7ee96977c4f12cbbed9366fcada87832c708b6/hd44780_helpers/src/lib.rs#L18 , I moved that function into its own crate.

chevdor, 2022-06-29:
@rene0 thank you for reporting this issue and researching on the topic. You will find your findings implemented in PR #40.

chevdor, 2022-06-30:
Closed by #40

closed 2022-07-05 by JohnDoneth