JohnDoneth/hd44780-driver #57: could not find `memory_map` in `hd44780_driver`

WuFengSheng, 2024-09-27:
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_hal::{
	clock::ClockControl, delay::Delay, gpio::Io, i2c::I2C, peripherals::Peripherals, prelude::*, system::SystemControl,
};
use hd44780_driver::{memory_map::MemoryMap1602, setup::DisplayOptionsI2C, HD44780};
use log::{error, info};

```

```
use hd44780_driver::{memory_map::MemoryMap1602, setup::DisplayOptionsI2C, HD44780};
  |                      ^^^^^^^^^^                 ^^^^^ could not find `setup` in `hd44780_driver`
  |                      |
  |                      could not find `memory_map` in `hd44780_driver`
```

JohnDoneth, 2024-09-27:
It looks like this is due to https://github.com/JohnDoneth/hd44780-driver/pull/54 (cc @ColinTimBarndt).

ColinTimBarndt, 2024-09-28:
Yes, you're depending on an older version of this library. Change your dependency to use this git repository directly, that should fix it.

ColinTimBarndt, 2024-10-03:
@WuFengSheng try this:

```toml
[dependencies]
hd44780-driver = { git = "https://github.com/JohnDoneth/hd44780-driver.git", rev = "99152ccfe2263e70c480f7d1f417181b00423d37" }
```

The `rev` commit hash is optional, but I recommend it so that your code doesn't suddenly break when changes are made in this repository.